mirkix/ardupilotblue #3: ArduPlane blue?

dacobi, 2017-05-23:
Hi

Just got my BeagleBoneBlue and followed your guide to setup ardupilot, but I can't find arduplane?

Am I doing something wrong or is it still in development or...?

Kind Regards
Jacob

mirkix, 2017-05-23:
@dacobi Hi Jacob,

you have to compile ArduPlane by yourself, follow https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md#cross-compile-ardupilot-faster-with-ubuntu-computer but with:
```git checkout ArduPlane-beta```
and
```./waf configure --board=blue```

I can send you a ArduPlane bin file as well, just PM me.

Regards
Mirko

Kimo-k-o, 2017-05-23:
Thank you Mirko for all of your help on walking us through this. I have been trying to build a working arduplane-blue, so far with only partial success. At this point I _am_ interested in getting the ArduPlane bin file from you, but it looks like PM has been removed from github...

dacobi, 2017-05-24:
I'd really like a copy of the binary but can't find where to PM you?

Regards
Jacob

mirkix, 2017-05-24:
My address is mirkix@gmail.com

Kimo-k-o, 2017-05-24:
Hi Mirko,

 

Thanks again for your generous contributions to Open Source development. I hope to do the same as my skills improve. 

 

I am working for the first time with Linux, and for the first time with ArduPilot.  Although I've done a lot of coding long ago, my career has been mostly in the development of opto-mechanical devices. 

 

At this time I'm pursuing my personal interests in AUV's. The slightly buoyant underwater craft I'm developing behaves like a twin-engine flying wing, with an inverted wing. My current strategy is to begin with the ArduPlane code and combine it with the appropriate ArduSub-related behaviors. 

 

I've assembled a breadboard with the various electronic hardware components and, after following your arducopter-blue recipe, the BBBlue can now talk to Mission Planner: ready for a choice of frames.  I spent several days trying to build the ArduPlane-beta for the BBBlue, but something isn't yet right… presumably on my end. 

 

I'll continue to battle against my Linux ignorance, but I'd greatly appreciate it if you could provide me with the bin file for installation. My email is limited to about 10M files, and I'm also new enough to GitHub that I don't know if files like this can be transferred there, so here's a link to my DropBox account <https://www.dropbox.com/sh/i1kawgcezhucv0v/AABazPOlhD0vaFEVQsoDX7GWa?dl=0> , if you want to use that.

 

All the best,

 

Tim

mirkix, 2017-06-01:
http://bbbmini.org/download/blue/

closed 2017-06-01 by mirkix