mirkix/ardupilotblue #5: arducopter fails to start with "No INS backends available" error.

braincore, 2017-08-14:
I followed the instructions provided. I'm running arducopter with no parameters and getting a "No INS backends available" error:

```
root@beaglebone:~# ./arducopter-3_5_0 
DataFlash_File: buffer size=16384
�       Q}��    Q��No INS backends available
�       Q� root@beaglebone:~# 
```
This is triggered by this line: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor.cpp#L815

This is unexpected since I would expect the logic for accessing the IMU over i2c to solve this: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor.cpp#L780

Is there anything else I can provide to help debugging?

mirkix, 2017-08-22:
@braincore Did you compile arducopter by yourself?

braincore, 2017-08-25:
@mirkix: I've tried using the binary you provided as well as compiling arducopter myself. Both produce the same error. Note `rc_test_imu` from Robotics Cape works just fine.

I've also verified that this backend is being used: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor.cpp#L781

OscarJGV26, 2017-09-03:
I am going to leave a comment here. I did exactly the same as brain and I am getting same result. Funny thing is, when I did all the process for the first time it look ok since it was showing something like  this every second or so... 

 Q}��    Q��       Q}��    Q��        Q}��    Q��      Q}��    Q��

I assume this is some kind of heartbeat to attempt communication with ground station.

OscarJGV26, 2017-09-05:
@braincore Hi brain,

Try my suggestions in ([1] https://github.com/mirkix/BBBMINI/issues/41). Particularly I think reinstalling the kernel from step 5 of ([2] https://github.com/mirkix/ardupilotblue) is what did the change. I think it may have been something with loading the wrong layout as in step 16 of [1] or choosing the wrong board while compiling in step 4 of [1].

Regards,
Oscar

braincore, 2017-09-11:
@OscarJGV26: I've always been compiling using `board=blue` and was never using the BBBmini dtb. Do you have any dtb set at all in `/boot/uEnv.txt`? I've also tried a fresh image using Debian Stretch, but also to no avail.

Any chance you have a better idea of what you changed?

OscarJGV26, 2017-09-11:
@braincore 
I was thinking that my post was not correct haha! ^.^

Well... for a start, I didn't do anything that is not on [1] or [2]... Basically I flashed a fresh image, then ran [1] without step 16 then [2] again, and then compiled ardupilot from [1] using board=blue. 

Another thing that I noticed once was that if I used any library from the robotics cape it stopped working... 

Other than that, I really have no clue and no I didn't set any special dtb in /boot/uEnv.txt...

braincore, 2017-09-11:
@OscarJGV26: Yes, I think that final observation is what I've correlated as well: "when I tested code made from robotics cape it stopped working."

I'll try a fresh install and avoid running rc.

OscarJGV26, 2017-09-11:
@braincore Yes, now that I think about it... That might be the reason.... Particularly if you run code to write to the prus... I think thats what messes it up.

But anyways, please update me since I am also trying to use robotics cape together with some ardupilot function... I'm now able to write PWM inputs with the ardupilot's pru's with this library (https://github.com/OscarJGV26/INPUTS). 

I also give a library to log data, read receiver (this is not currently working) and send some data through udp if you happen to have your own ground control station like me. 

I will also be making a library to called SENSORS to read the IMU/barometer/gps and I was thinking about using robotics_cape library for imu/barometer but oh well.. lets hope I can! :D

braincore, 2017-09-17:
@OscarJGV26: I've had limited time so I haven't investigated whether it has to do specifically with the PRUs. After doing a re-install and ignoring the rc libraries, and one hard reset, the INS backend is found. Lots of voodoo magic though, and I really don't understand what's going on.

I've connected with the APM Planner v2 ground station, but the radio calibration section cannot find the RC, and arducopter prints: "Throttle below Failsafe." So the debugging continues :)

OscarJGV26, 2017-09-17:
@braincore 

Haha, I was exactly like you man. Had to debug the hard way this last 2 weeks. Fortunately, I was able to set up just how I needed it and I'm able to fly now with my own code using part of the rc library (IMUs,barometer,Serial Coms, GPS) and part of the ardupilots PRUs. 

For your receiver to work you need to run this line at some point before you run ardupilot, either start up or whenever you want... 
/bin/echo pruecapin_pu > /sys/devices/platform/ocp/ocp:P8_15_pinmux/state

Check this out...
https://github.com/mirkix/ardupilotblue

braincore, 2017-09-17:
@OscarJGV26: It's great to have you around paving the way! haha. I'm interested in checking out the code you're writing once I get mine going. What are you trying to build/do?

Just to confirm with you, I didn't seem to have a `/sys/devices/platform/ocp/ocp:P8_15_pinmux` folder until I swapped the PRU overlay line in `/boot/uEnv.txt` to: `uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo`. Did you do this as well?

I have the folder now, and I changed the state value. No luck though. Any other tips?

OscarJGV26, 2017-09-18:
@braincore

I will share my code in my github soon if you want to look at it or use it. Unfortunately, it will not have the control laws because I can't upload that yet so you will have to use your own PIDs or whatever. If you have Labview, I also have a VI for receiving data through UART or UDP (I prefer UART)

As to what I am doing/building... I am trying to implement Nonlinear Model Adaptive Predictive Control in UAVs in general and I have a library for that that I will be publishing soon, but more specifically, I will be building a VTOL based on a Flying Wing/Tricopter/Tilt-Rotor Design of my own.

I removed ardupilot first because it was not running as rt (it runs with 13 priority - don't know why), and second because it uses almost 40% of the CPU, even without flying/gps/ultrasonic connected and third because its a mess to add libraries to it. Currently I am flying already in acro mode with 4% CPU (no data fusion), but I doubt any EKF will take more than 10-15% extra, specially if it doesn't have GPS. The thing is, I really need the full CPU to myself because the optimization requires quite a lot of power. Plus, I will be doing some image processing with OpenCV as well, so the more tailored is my code the better.

As to any of the modifications that you mention, it's like I've said before. I didn't do anything that is not in [1],[2]. In fact, you should check my RECEIVER or INPUTS library. You can test both of them without even installing ardupilot fully. Also, make sure you are powering your receiver correctly and using the correct pins.

braincore, 2017-09-20:
@OscarJGV26: Solved! I've been using the DSM2 port, rather than E4 pin 4. Now that I've got it all working from a software side, I need to work on the design of the 3d printed frame as it's too heavy.

After that, I plan to implement my own control algorithm. Looking forward to whatever you share, and time permitting, I hope to get some of my own code out there.

OscarJGV26, 2017-09-20:
@braincore 

Good. Time to get on to it then! ^.^

Well, if you want you can try this template (https://github.com/OscarJGV26/Template) but BE AWARE that it uses the robotics cape. XD

It basically acquires receiver, performs imu ekf for attitude estimation, writes pwm signals to motors/servos (or angles if you want to write angles angles), logs data into a file which could be your SD card mounted (in my case /media/SD) and sends data through UART and UDP in case you have a ground control station receiving. I have mine with Labview so if you are into labview, I can share that as well...

Notes: 
-For the EFK you have Q_info={q_cov_noise (all axis), ignore, ignore}, R_info={acc_noise,mag_noise}. 

-I do some strange correction of the order and orientation of the axis because that is how I have it in my system, e.g. roll/pitch/yaw instead of pitch/roll/yaw of robotics cape. I might be correcting that soon.

-The accelerometer calibration currently only calibrates the horizontal plane.

-You can put your control algorithms in the float motors[4]={0,0,0,0} section... I have channel 5 for safety and channel 6 for flight mode selection...

-My code is ready for CPPM receiver only. So, if you have DSM or Sbus you will have to change the process_pulse function probably and maybe even call the function with some timers or something but this can be easily done checking the process_pulse function of ardupilot.

Anyways, hope you can get going now. 

All the best!

closed 2018-01-04 by mirkix