mirkix/ardupilotblue #6: Modify Arducopter firmware/coding

OscarJGV26, 2017-09-02:
Hi,

I would like to modify arducopter firmware and coding for implementing my own control loops. 

Could you give me an idea where to start?

Regards,
Oscar

OscarJGV26, 2017-09-04:
Okay, I've got ardupilot installed now and I've been reading about the ardupilot code base in (http://ardupilot.org/dev/docs/learning-ardupilot-the-example-sketches.html#). I am trying to understand it implementing some of the examples there.

Could someone tell me what would I need to do to simply get the RC signals from the PRUs and send ESC (PPM or PWM) signals to the PRUs, without the trouble of the all the HAL libraries... 

I just neeed something very simple and I am making my own code. I don't want to get through all the trouble of understanding the whole ardupilot code. 

Regards,
Oscar

mirkix, 2017-09-26:
@OscarJGV26 Hi Oscar, you can check this https://github.com/ArduPilot/ardupilot/blob/master/Tools/Linux_HAL_Essentials/pru/aiopru/RcAioPRUTest.c

mirkix, 2017-09-26:
or this https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/examples/RCOutput/RCOutput.cpp

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/examples/RCOutput2/RCOutput.cpp

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/examples/RCInputToRCOutput/RCInputToRCOutput.cpp

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/examples/RCInput/RCInput.cpp

OscarJGV26, 2017-09-26:
Thank you @mirkix 

 I asked about the "automatic" update cause you have to service the PRUs of the robotics cape library, so I was really looking for something more like yours and I've already been able to code everything as I wanted outside ardupilot with your PRU codes which work perfect as expected.

Regards,
Oscar

mirkix, 2017-09-26:
@OscarJGV26 Sound good, so I can close this issue?

OscarJGV26, 2017-09-26:
@mirkix Sure. Delete it if you want... It's not even an issue but tbh I didn't know gitter before this so this was the wrong place to post this kind of questions. 

Regards.

mirkix, 2017-09-26:
@OscarJGV26 I will close the issue.

closed 2017-09-26 by mirkix