mirkix/ardupilotblue #11: Missing information

ejhaze, 2018-01-06:
Hi Mirkix! 

I'm not sure if this is the right forum for this. I have followed these directions over and over for the last 4 months, so this is not without really trying to make your directions work. My issue comes when trying to access Mission Planner, the MAVlink connection always times out. I guess the problem arises somewhere between :
**Prebuild ArduPilot** and **Run ArduPilot**. There must be missing instructions. 

I would like to do the "Compile ArduPilot native on BeagleBone" like you did on the BBBMini, but the checkout function responds saying that git checkout ~ is not a git repository. 

Nothing addresses the "blue-arduBUILD" naming function that is rendered following these directions. 

But as an aside, it would be nice to either change the run ardupilot to : 
sudo /usr/bin/ardupilot/blue-arducopter  

or show the step how you got to: 
sudo /home/debian/arducopter

To add a step, that these directions are missing (from the bbbmini page - modified): 
"scp /usr/bin/ardupilot/* scp  debian@beaglebone:/home/debian/"

I've never posted on one of these, but maybe if you could clarify some of these points, it would make life easier for someone else. Also, great pic of where to put the E4 pin. I had to hunt that down here: 
http://esys.ir/images/img_Item/1317/Files/BeagleBone_Blue_sch.pdf

Vielen Dank fuer Ihre Hilfe! 
EJH

mirkix, 2018-02-11:
@ejhaze What command do you exactly use when the error from git comes?

ejhaze, 2018-02-11:
Hi @mirkix, 

After further tinkering, I was finally able to get the BBBlue to talk to Mission Planner/ArduPilot (-C tcp:192.168.7.1:1243:wait worked for me). 

The pre build of Ardupilot from your link has never worked for me. I've ssh to my BeagleBone and it does nothing. 

I did use @jadonk's way of apt-get install -y ardupilot-copter-blue to get ardupilot on the BBBlue, but the receiver was still not recognized within Mission Planner.

Currently, I'm following the steps for the BBBMini to compile ArduPilot native to the BeagleBone. This is to be able to follow @rflowers steps on configuring the PRU on the other open issue. This allows access to the HAL Linux Essentials. 

With current directions, I am also seeing problems with ArduPilot not recognizing the DSM2 or the eQEP4. However, on the Beaglebone, when I run rc_test_dsm, the BBBlue recognizes that my transmitter is sending signal to the receiver. 

I was able to have ArduCopter start on boot, which was a big step, but I still don't have the control through Mission Planner when I get in there. That goes for the I2C, GPS (for uBlox-M8N), the DSM2 (Satellite Receiver), the eQEP4 (radio Receiver), and anything that is not through the USB0 serial. 

Again, at this point this may be bleeding into the other open issue. 

Thanks again! 
EJH

ejhaze, 2018-02-17:
This is interesting. I reflashed my BBBlue and started the process over, yet again. But this time, before adding anything on the uEnv.txt to configure the GPS/I2C/UARTs and other pins, I just ran: 

root@beaglebone:/opt/scripts/tools# sudo ./version.sh

Check out the pin errors towards the bottom. Any suggestion on how to get around that conflict? 

git:/opt/scripts/:[89f2e9309c2322c15e2bb6b55afd214313d94842]
eeprom:[A335BNLTBLA21712EL003950]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g60ae27e116]:[location: dd MBR]
kernel:[4.4.113-ti-rt-r146]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_overlays=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
**[    1.535830] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.580072] pinctrl-single 44e10800.pinmux: pin 44e10954.0 already requested by 48024000.serial; cannot claim for ocp:P9_21_pinmux
[    1.592058] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P9_21_pinmux) status -22
[    1.599500] pinctrl-single 44e10800.pinmux: could not request pin 85 (44e10954.0) from group pinmux_P9_21_uart_pin  on device pinctrl-single
[    1.621020] pinctrl-single 44e10800.pinmux: pin 44e10954.0 already requested by 48024000.serial; cannot claim for ocp:P9_21_pinmux
[    1.632822] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P9_21_pinmux) status -22
[    1.640259] pinctrl-single 44e10800.pinmux: could not request pin 85 (44e10954.0) from group pinmux_P9_21_uart_pin  on device pinctrl-single
[    1.669707] pinctrl-single 44e10800.pinmux: pin 44e10950.0 already requested by 48024000.serial; cannot claim for ocp:P9_22_pinmux
[    1.681513] pinctrl-single 44e10800.pinmux: pin-84 (ocp:P9_22_pinmux) status -22
[    1.688950] pinctrl-single 44e10800.pinmux: could not request pin 84 (44e10950.0) from group pinmux_P9_22_uart_pin  on device pinctrl-single
[    1.710470] pinctrl-single 44e10800.pinmux: pin 44e10950.0 already requested by 48024000.serial; cannot claim for ocp:P9_22_pinmux
[    1.722268] pinctrl-single 44e10800.pinmux: pin-84 (ocp:P9_22_pinmux) status -22
[    1.729705] pinctrl-single 44e10800.pinmux: could not request pin 84 (44e10950.0) from group pinmux_P9_22_uart_pin  on device pinctrl-single
[    1.761167] pinctrl-single 44e10800.pinmux: pin 44e10984.0 already requested by 48022000.serial; cannot claim for ocp:P9_24_pinmux
[    1.772980] pinctrl-single 44e10800.pinmux: pin-97 (ocp:P9_24_pinmux) status -22
[    1.780419] pinctrl-single 44e10800.pinmux: could not request pin 97 (44e10984.0) from group pinmux_P9_24_uart_pin  on device pinctrl-single
[    1.802142] pinctrl-single 44e10800.pinmux: pin 44e10984.0 already requested by 48022000.serial; cannot claim for ocp:P9_24_pinmux
[    1.813943] pinctrl-single 44e10800.pinmux: pin-97 (ocp:P9_24_pinmux) status -22
[    1.821379] pinctrl-single 44e10800.pinmux: could not request pin 97 (44e10984.0) from group pinmux_P9_24_uart_pin  on device pinctrl-single
[    1.850831] pinctrl-single 44e10800.pinmux: pin 44e10980.0 already requested by 48022000.serial; cannot claim for ocp:P9_26_pinmux
[    1.862637] pinctrl-single 44e10800.pinmux: pin-96 (ocp:P9_26_pinmux) status -22
[    1.870075] pinctrl-single 44e10800.pinmux: could not request pin 96 (44e10980.0) from group pinmux_P9_26_uart_pin  on device pinctrl-single
[    1.891794] pinctrl-single 44e10800.pinmux: pin 44e10980.0 already requested by 48022000.serial; cannot claim for ocp:P9_26_pinmux
[    1.903595] pinctrl-single 44e10800.pinmux: pin-96 (ocp:P9_26_pinmux) status -22
[    1.911032] pinctrl-single 44e10800.pinmux: could not request pin 96 (44e10980.0) from group pinmux_P9_26_uart_pin  on device pinctrl-single
[    1.946541] pinctrl-single 44e10800.pinmux: pin 44e10990.0 already requested by ocp:P9_31_pinmux; cannot claim for 481a0000.spi
[    1.958105] pinctrl-single 44e10800.pinmux: pin-100 (481a0000.spi) status -22
[    1.965284] pinctrl-single 44e10800.pinmux: could not request pin 100 (44e10990.0) from group pinmux_P9_31_spi_pin  on device pinctrl-single**

silver2row, 2018-04-08:
Hello,

This may help: http://ardupilot.org/dev/docs/building-for-beaglebone-black-on-linux.html.

Seth

mirkix, 2018-04-08:
@silver2row Your link is complete out of date.

silver2row, 2018-04-09:
@mirkix You are right. I see they have little for compiling on BBB and the variations. I am glad you started it.

Seth

devtodev, 2019-09-13:
Hi! I am making an open source rover with beaglebone blue, now I am looking to connect the M8N GPS I bought, but the GPS connectors are for the Pixhawk, so I am looking for cut the cables and soldering, but... for what I saw the GPS connector on the beaglebone have 5v. And my asumption is the UART on GPS beaglebone connector is 5v too.

How you wire the M8N GPS to the beaglebone? Thanks!