mirkix/ardupilotblue #12: Please provide libraries you used/created

Thalaivar, 2018-01-28:
A wonderful port to blue for ArduPilot, I request you to please provide the libraries you made/used that access the peripherals of the blue/BBBMini

mirkix, 2018-01-28:
@Thalaivar You can find all the software in the ArduPilot source code. 
https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_Linux
https://github.com/ArduPilot/ardupilot/tree/master/Tools/Linux_HAL_Essentials/pru/aiopru

Thalaivar, 2018-01-29:
Those libraries are heavily interlinked right? Is there any way I can use them as independent libraries for my own program?

Thalaivar, 2018-01-29:
Those libraries are heavily interlinked right? Is there any way I can use them as independent libraries for my own program? 

I used arm-linux-gnueabi-g++ to compile GPIOTest.cpp and put the lib directory as ~/ardupilot/libraries

I got these errors:
[image]

juvinski, 2018-01-29:
How are you configuring and compiling ?

Thalaivar, 2018-01-29:
Apologies, but I don't understand what you mean by "configuring". As I mentioned, I simply set the arm-linux-gnueabi-g++ to search for libraries in the _**ardupilot/libraries**_ folder and compile for C++11.

Please let me know what the possible problem can be...

juvinski, 2018-01-29:
You need to use the waf utility.
./waf configure --board=bbbmini 
You you want to compile using make/gcc you need to include the libraries directory

Thalaivar, 2018-01-29:
Yes but suppose I wanted to write a imu_read.cpp using AP libraries, I can't build with waf right? How would I write the wscript for it?

juvinski, 2018-01-29:
Hi, you can use waf.
You need inside the vehicle you want change the wscript file to include your sensor folder.

Thalaivar, 2018-01-29:
The problem with that is, ardupilot expects all of the code to be in a set form where, for a vehicle, you have vehicle.h, vehicle.cpp and ArduVehicle.cpp where

Thalaivar, 2018-01-29:
I do not understand how th HAL is structured. I don't know how the AP_HAL_MAIN() callback works. Hence I just want to use them as normal c libraries without all the HAL  call back etc. How can I do that?

closed 2018-01-30 by Thalaivar