mirkix/ardupilotblue #21: A question more than an issue

crazyquark, 2018-10-09:
Hello,

I am new to the Beaglebone Blue and I don't understands its power circuit.
I am trying to figure out how to connect and power using a LiPo batter a Racestar 4in1 ESC.
You say in the README that there is no need to enable power via the motors header for ESCs.
Then, how will the ESC be powered? From that tiny power out JST-SH connector on the board? I have a hard time imagining how to connect such a fragile socket to the beefy wires coming out of the ESC.

Also, If I understand correctly one is supposed to only use the BBBlue to charge the battery and power will be provided somewhere different?

Sorry if I am being a total noob but a diagram of the power scheme for such a drone would help me tremendously. Sadly I couldn't glean this info from the picture of the assembled drone you posted.

The ESC I am trying to use is this one: https://www.banggood.com/Racerstar-RS20Ax4-20A-4-in-1-Blheli_S-Opto-ESC-2-4S-Support-Oneshot42-Multishot-for-FPV-Racer-p-1068210.html?

suz1, 2018-10-09:
Hi, if I can assist. You need to power the motors separately from the beaglebone, but with a common ground. The Electronic Speed Controllers need the main power from your lipos, and the pwm signals from the bbb. If you accidentally form a circuit through the bbb from the full lipos you will fry it, as the motors require more current than the board can handle. The motor connectors on the bbb are not used in its application of ardupilot. You either power the bbb with a separate lipo or take it carefully from the main lipo with necessary protections in place. Hope this helps.

crazyquark, 2018-10-09:
That makes a lot of sense, thank you. So some sort of BEC is needed.

suz1, 2018-10-10:
If you want to use servos rather than ESCs, the beaglebone blue doesn’t by default put out enough power through the pwm connectors as ESCs have a separate power supply. You need to enable servos in software and supply sufficient power in the correct place (ie servos won’t work properly if you are just powering it through a USB port). One option is to use the bbb lipo to supply power. I trust you’ve inspected the bbb schematics, this appears to be a simplified section of the battery servo power circuit (servo_en is the pin enabled in software).