mirkix/ardupilotblue #25: Frsky Telemetry with BB Blue

ninjatrent, 2019-10-15:
I would like to use the Frsky Taranis QX7 Transmitter with Flight Deck Software. I have a Frsky X8R Receiver S Port Telemetry serial output connected to the uart5 port on the BB Blue and the SBUS pin from the QX7 receiver connected to the SBUS on BBB. 

In the Mission Planner software under Full Parameter Tree, I have changed the Serial 5_PROTOCOL value to "10" in order to enable Frsky SPort Passthrough. The transmitter is detected and can be calibrated with Mission Planner. 

The Flight Deck Software on the QX7 Transmitter does not detect the Flight Controller (BBBlue). There is also a telemetry radio module connected to uart1 on BBB with the USB telemetry module connected to the GCS. I have edited the /etc/default/ardupilot file to

TELEM1="-C /dev/ttyO1
TELEM2="-A udp:<GCS IP address>:14550"
GPS="-B /dev/ttyO2"
Unnamed="-F /dev/tty05"

Do you have any suggestions for using the Flight Deck software on the Frsky Taranis QX7 Transmitter with the Ardupilot Blue image for the BB Blue?

mirkix, 2019-10-20:
@ninjatrent Hi, FrsSky Telemetry will not work out of the box. You have to invert RX and TX and put them together

ninjatrent, 2019-10-20:
My apologies for I don't understand. Can this be done by connecting RX to RX and TX to TX on BBB and the RX and TX wires connected?

closed 2020-01-21 by ninjatrent

MahmoudElShazly, 2020-06-18:
Did you get it working? did you figure out the connections?

ninjatrent, 2020-06-18:
No sir. I was unable to find a solution to this issue.

MahmoudElShazly, 2020-07-11:
Hi @mirkix , here is what I did
0. Mapped Uart 5 in beagle bone to Serial 5 in Ardupilot 
1. Configured Serial 5 to 10 'Passthrough'
2. Inverted Tx and Rx from within Ardupilit
3. Connected the Tx wire to the Signal Pin in the S-Port in R9 Stab receiver
4. Connected Tx and Rx together (short circuit)

Then in the transmitter I started 'Discover New Sensors' but no data came.
Do I miss anything?

dziedzic99, 2022-02-28:
Any progress on this?

dziedzic99, 2023-03-11:
@RedgarTech Response time from BBB is simply too long and it times out. Probably due to the OS standing in-between. Having some more experience since a year ago when I tackled this matter I highly recommend switching to ELRS and moving away from Frsky.

RedgarTech, 2023-03-11:
@dziedzic99 here's the thing, I found someone here
 [https://discuss.ardupilot.org/t/connecting-rx6r-or-r-xsr-to-beaglebone-blue/91963](Frsky Receiver Ardupilot Discussion)

who found a way for it to work. I am attempting to get it working now actually.

RedgarTech, 2023-03-11:
> @RedgarTech Response time from BBB is simply too long and it times out. Probably due to the OS standing in-between. Having some more experience since a year ago when I tackled this matter I highly recommend switching to ELRS and moving away from Frsky.

Also, did you eventually get Frsky working?

juvinski, 2023-03-11:
Hi,

I have sucess on this, but with black. I detected some issues with receivers, Beagle ports demand at least 3.0 volts do be detected as high, so the Inverted circuit I'm using is generating 3.3 volts and is working fine.
This for PRU too - I have several R161 receivers and this is generating sbus signal with 2.65 volts and this is not detected by the beagles, I had to use a shift level circuit to increase to 3.3 volts to get this working too.